zynthian/zynthian-webconf #118: Snapshot and Midi options

mheidt, 2020-01-19:
As you might know, the MIDI options are saved in the snapshot file as well.
This is a nice feature, but can be troublesome.
I play in different bands with different tuning (442 and 440)
I reckon, if I change the Midi profile, the new setting will be taken.
It’s not nice if you have to change the midi profile forth and back just to get the right tuning.
Besides that I have a screenless zynthian also and cannot change the midi profile just like that.

I would need a midi/zynthian-ui option, that says “Always use global Midi settings” or “Ignore Midi settings in snapshot file”
Or should we have a checkbox in each snapshot?

The other option could be to not save SINGLE_ACTION_MODE, master channel, tuning.

Another alternative would be having the option Snapshot in the MidiProfile selection.

jofemodo, 2020-01-20:
Hi @mheidt !

Good work! I like it ... but i would change the labels. I would add:

+ Layouts => for the first text box with the layer info 
+ Options => for the rest, and will remove the  "MIDI Key / value" that is quite confusing.

Regards,

mheidt, 2020-01-20:
I added bootstrap-table to the system.
Because of https://github.com/zynthian/zynthian-webconf/issues/115 I want to change the Layer textbox to a table as well.
So I will remove table headers...

mheidt, 2020-01-21:
I created a new branch. If you have the time, you could check before I implement the possibility to add new options by selecting a midi profile.
I would add a + button and  a select box in the table footer.

mheidt, 2020-01-23:
But there is still the problem of the table width. 
Try to add long Midi Filter Rules.
The Trash-Icon is not visible anymore in that case

mheidt, 2020-01-24:
And there is a \n issue. I  am not quite sure, how to solve that.
Originally Ports has three lines. When I press + to import from default, the three lines are in one with \n.
Did you replace the \n in the snapshot file by <br>?

mheidt, 2020-01-24:
Ah, it's only due to js replacement of the code..the file is ok..I can handle that...but still...what about long midi rules...

jofemodo, 2020-01-24:
What about this?

https://css-tricks.com/almanac/properties/w/word-break/

mheidt, 2020-01-24:
yes, i will take a look into it.
But back to the carriage return..
The difference is, that with the normal load, the json came with the template rendering.
But when using the buttons, REST and self.write() is used.
Then, a double-backslash-n is returned. I could hack it and replace the double-back-slashes.
But I liked it better, if the REST return would be correct...I don't know how...

mheidt, 2020-01-24:
Regarding css.
The problem is, that CH#0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15 is to long already.
And this is nothing, word-break would break.
and rgd js/carriage return...I gave up and hack it per value:
.replace(/\\n/g,"\n")

mheidt, 2020-01-24:
hm, now word-break works. dunno, what I did wrong before.
Now I am happy...will close

closed 2020-01-24 by mheidt

jofemodo, 2020-01-24:
Great! Now it's perfect !! :-)