zynthian/zynthian-webconf #123: Search preset in all engines

mheidt, 2020-02-16:
Add new input to search for presets in all engines

jofemodo, 2020-02-17:
Please, could you elaborate a little bit more? ;-)

mheidt, 2020-02-17:
Below the Musical Artifact Search I will put a search field and button.
When submitted, all engines will be searched if a preset contains the search term like "harp".
Maybe I have to use a socket to return the result continuously like the logs.
Using zynapi calls of course.

mheidt, 2020-02-17:
Ok, I made the first test. Like expected it would take forever. Especially because of the 
if engine_cls == zynthian_engine_jalv:
	engine_cls.init_zynapi_instance(engine_info[0], engine_info[2])

So I have to use the websocket.
But how should I display the results?
Just a list or a tree?
Using the existing tree? But then the input field for the term should be in the top row.
And if something is entered, the select with the engines need to be disabled and the tree contains the engines as well.
Quite complicated everything.

Your opinion?

jofemodo, 2020-02-18:
I would like to "cache" LV2-presets in the same way plugins are cached. The cache would be regenerated when new plugins/presets are installed. This would speed up the loading time for the most important plugins, like Dexed or OBXd, also will speed up the preset manager, including the global search we want to implement. So ... let's wait a little bit for implementing this task, until cache is implemented. I hope very soon ... ;-)

jofemodo, 2020-03-02:
Cache is now implemented and global search should be a lot faster now ;-)

mheidt, 2020-03-04:
Definitely was worth the wait :)
I committed something in gh123.
Where do we go from here?

mheidt, 2020-03-05:
I was thinking that you could be able add the found presets to snapshot.
I see following open issues:
1. Pianoteq is not integrated, because we couldn't change presets anyways so far. That's why Pianoteq is not in the result list either.
2. How do we connect the Snapshot with the Preset area? 
Proposal: 
a) The Snapshot area gets a CreateEmptySnapshot (note to myself: is the ui action a class method and in zynapi already?)
b) When you select a snapshot, you can mark it with a star, which means that it is selected globally. Don't how we should call it, so that the user understands it.
c) When a snapshot is globally selected, I could add a midi channel select box (1-16) in front of the search result in the preset area.
At the bottom there is a button which says, that this preset will be added as new layer with channel XY in the globally selected snapshot.

riban-bw, 2024-03-12:
It feels like this is complete. I will close this ticket. If it needs to be reopened, please also move to issue tracker.

closed 2024-03-12 by riban-bw