zynthian/zynthian-webconf #124: Bug report message length overflows webconf window.

wyleu, 2020-02-17:
**Is your feature request related to a problem? Please describe.**
The logging feature trims lines but there is no scroll mechanism, as this is fault finding this should not be dismissed as purely  cosmetic

**Describe the solution you'd like**
Whatever suits. Coloured word wrap has impressed me in the past, but a scroll bar would probably be pretty sufficient.

**Describe alternatives you've considered**
Watching listings in ssh shell: Requires user to know how to do ssl, and since this is a log report the people who might be investigating might have a limited system because they are bug hunting. . .

**Additional context**
[image]

mheidt, 2020-02-17:
Like that?

jofemodo, 2020-02-18:
I just tried it and the x-scroll is not very usable when log grows up. I think it would be better to use line wrapping.

jofemodo, 2020-02-18:
OK! I improved a little bit by adding the y-scroll to the log div with auto-resizing. Copied from midi-log ;-)

closed 2020-02-18 by jofemodo

mheidt, 2020-02-18:
I used the right and left keys btw. works perfect to me