zynthian/zynthian-webconf #126: Delete layers from snapshot

mheidt, 2020-03-01:
Why not a generic “delete layer” options that allow to delete any layer from the snapshots?

Also, you could add a checkbox to the MIDI profile. Something like:

[X] Exclude MIXER settings from Snapshots

And maybe a Add Mixer Layer that adds/replaces the current mixer layer

mheidt, 2020-03-01:
Please check and integrate the new ZYNTHIAN_MIDI_EXCLUDE_MIXER_FROM_SNAPSHOT

branch gh126

mheidt, 2020-03-01:
In case that you want an Add Layer, I need a zynapi call.
One that delivers the layer content. I can place it into the snapshot file.

jofemodo, 2020-03-02:
Why the 2 first layers (rows) do include the controller values?

mheidt, 2020-03-02:
2 is a bug. I only wanted the MIXER layer show the volumes, so that the row isn't too empty.

mheidt, 2020-03-03:
But it would be solved, if you would make the channel 1-based internally like planned.

jofemodo, 2020-03-03:
No no ... i never said this, my friend!
Internally, MIDI channel will be 0-based forever and ever. It has no sense to do it in other way.
I have to change it in the MIDI-rule parser, that can be considered "user-side", but again, internally the MIDI channel will be  0-based too, although it will parse 1-based rules.

mheidt, 2020-03-03:
Ah, ok. I thought it would make sense especially since mixer is 0 (ok channel=empty in some cases, but 0 for the user)

mheidt, 2020-03-03:
please check again

jofemodo, 2020-03-04:
I would remove the "0#" for the layers with MIDI channel set to "None".
Also take into account that the Mixer is not the only layer that can have midi channel set to None. For instance, MOD-UI layers also have midi channel set to None ...

mheidt, 2020-03-04:
Like that?
[image]

jofemodo, 2020-03-04:
Yes, like that! ;-)

jofemodo, 2020-03-04:
No man! It still showing the controller values for the MOD-UI layer ... and it's a mess because there are a lot!!

mheidt, 2020-03-04:
Damn, I loaded ModUI just for testing...never use it...but I didn't load a preset. maybe that's why it was empty in my case.
Try again.

jofemodo, 2020-03-05:
Merged!

closed 2020-03-05 by jofemodo