zynthian/zynthian-webconf #131: Add links to dashboard-block

wyleu, 2020-04-14:
**Is your feature request related to a problem? Please describe.**
No it's an ease of use request. Links from the dashboard-block objects would save hamburger menu selection 

[image]

**Describe the solution you'd like**
An anchor tag pointing to the appropriate selection ( Hardware, MIDI etc.

**Describe alternatives you've considered**
Extra key presses. I suspect there are some uses coming to this for the first time, and as the entry point to the eco-system it would be nice to proactively acknowledge activity by new users by responding to touch. But that could be just me.

wyleu, 2020-04-14:
It's a small thing. IN a previous version you could press on the title bar to access the submenu. IF its been decided to drop it no problem.

closed 2020-04-14 by wyleu

jofemodo, 2020-04-14:
Sorry @wyelu! This feature has not been implemented in the past. I'm pretty sure of that.
It has no sense because i don't know where to point the link in most of cases. Only MIDI and perhaps NETWORK could have a clear link, rest of titles are should open drop-menus from the topbar and this is not very nice.