zynthian/zynthian-webconf #132: Feature: Ability to copy zynthian configuration information from webconf

riban-bw, 2020-04-14:
**Is your feature request related to a problem? Please describe.**
Each time I report a bug I need to manually copy various data from webconf to github. It would be advantageous to be able to copy this info in a single operation.

**Describe the solution you'd like**
A button or link or menu option to populate the copy buffer with information required for bug reports.

**Describe alternatives you've considered**
It may be advantageous to have a link that opens a bug report with this data prepopulated.

**Additional context**
Required information:
**HARDWARE**
- Raspberry Pi Model 
- Soundcard
- Display
- Wiring
- GPIO Expander

**SYSTEM**
- Raspbian version
- Build Date
- Memory

**SOFTWARE**
- zyncoder
- zynthian-ui
- zynthian-sys
- zynthian-data:
- zynthian-webconf

jofemodo, 2020-04-14:
We have several options here:

+ copy to the clipboard => quick, easy and convenient
+ screenshot from the dashboard => how to enforce/ease this?
+ open an issue from the webconf, by opening a new window and  filling some fields. We would need to use github API ... 

@mheidt, what do you think about this?

mheidt, 2020-04-14:
duplicate to
https://github.com/zynthian/zynthian-webconf/issues/108
https://github.com/zynthian/zynthian-webconf/issues/71

closed 2020-04-14 by mheidt