zynthian/zynthian-webconf #133: Post lib-snapshot/ajax/save return Error 500

le51, 2020-05-14:
While renaming a snapshot:
**Request:**
URL :http://192.168.1.30/lib-snapshot/ajax/save
Method :POST

**Parameters:**

NEW_BANK_NUM | 1
-- | --
SEL_FULLPATH | /zynthian/zynthian-my-data/snapshots/000/004-ZY+>+Cris+Owl+Alvarez>ambient+pad+3.zss
INSTALL_FPATH |  
SEL_NAME | ZY+>+Cris+Owl+Alvarez>ambient+pad+3
SEL_BANK_NUM | 000
SEL_BANK | 000
SEL_PROG_NUM | 004

**Response:**

500 Internal server error

But processing is done correctly (file is renamed)

jofemodo, 2020-05-14:
It should be fixed by https://github.com/zynthian/zynthian-webconf/commit/122ddd85bc70c6e0d26023a5acc7748b814a06aa.

Please, update and test ...

le51, 2020-05-15:
It works correctly now. Thanks.

closed 2020-05-15 by jofemodo