zynthian/zynthian-webconf #135: WAV & MIDI File upload from Webconf

wyleu, 2020-05-19:
**Is your feature request related to a problem? Please describe.**
Audio & recordings can be downloaded from the zynth via the webconf interface. 
However new wav and MIDI  file samples cannot be uploaded to the zynth

**Describe the solution you'd like**
A file picker interface that allows multiple WAV and MIDI files to be uploaded to the banked structure of the zynthian.

**Describe alternatives you've considered**
Uploading via sftp to appropriate directory.

**Additional context**
Obviously these are external files and should be treated with great suspicion!
Check for magic numbers and file extension at least? 
How much meta data could we or should we display about files?
Length, sample rate. .

mheidt, 2020-05-19:
this new mutagen threw an exception for some wavs, I uploaded.
That's why I changed the try/except

Is zynthian-ui coping with mp3? The webconf doesn't have the folder yet.

jofemodo, 2020-05-19:
Hi @mheidt!
Thanks for adding the feature! You are fast ... jejeje!

Mutagen's support of wav files is very recent, so could be some bug. Please, report to the mutagen github.

Regarding UI, yes, it supports wav, ogg and mp3 files, so mp3 folder should be added to webconf.
Also, i noted that only wav uploads are active by now, why didn't include ogg files too?

Finally, please, move the upload button to the right area, same than the other library tools. When the list grows up, the button is hidden  ...

mheidt, 2020-05-20:
Ogg should work as well. You need to select the folder. This changes the filter.
I didn't take the right div, because it is invisible when no capture is selected. I would need a new div hierarchy.

mheidt, 2020-05-20:
try again

jofemodo, 2020-05-21:
OK! It seems to work fine now ...
Are you OK with closing the task and presenting the new feature on the forum? ;-)

wyleu, 2020-05-21:
:-D

closed 2020-05-23 by jofemodo