zynthian/zynthian-webconf #142: Show verbose description of LV2 plugins

riban-bw, 2020-07-10:
**Is your feature request related to a problem? Please describe.**
webconf provides a mechanism for enabling and disabling LV2 plugins. The list shows the plugin name, type, URL and enable state. The name is often cryptic and hence it may not be obvious what each plugin is  does.It would be advantageous to display a more verbose description of the plugin.

**Describe the solution you'd like**
Within each LV2 ttl there may be a rdfs:comment which may provide a verbose description of the plugin. If this is available it would be good to display it within the webconf. This may prove too verbose, making the layout unacceptable so maybe this description is truncated to a single line with ellipsis and ability to expand to show rest and / or tooltip which shows when pointer hovers over row / name / description. Maybe the URL could be turned into a hyperlink on the name, freeing up the URL column to show the description.

jofemodo, 2020-07-10:
Currently, only LV2 cached info is shown by the webconf LV2 manager. De LV2 is not currently being cached, and doing so would increase the cache size by x10, so i think it's not a good idea. But we could implement a AJAX action that would query the LV2 world and get the description for the required plugin. It's not difficult at all and it would solve the problem ;-)

riban-bw, 2024-03-12:
There is now a display of processor description in the UI. We could populate this with the rdfs:comment if available **and** not overwritten manually.