zynthian/zynthian-webconf #143: Report the presence, size and full % of an attached memory stick

wyleu, 2020-08-05:
**Is your feature request related to a problem? Please describe.**
If you are using the Audio recorder to record it is preferred to use a USB stick, but from the webconf it is not easy to tell if there is one plugged in and how much available space there is on it.

This could be reported on either or both the home screen and the capture screen. ( Record button ????) 

For instance, this rig can be accessed remotely to allow recording, but the webconf doesn't report available space. One has to log on to the console.

[image]

**Describe the solution you'd like**
A report of whether or not the USB stick is inserted, and registered. This works as an excellent health check of the stick itself, which can be 'suspect' which is not something one wants to report after the performance.

An  overall capacity of the stick

An overall usage figure of the USB stick

 a percentage of usage 

an available record time of the stick.

**Describe alternatives you've considered**
Logging on to the ssh console and examining the stick using df -h

**Additional context**
Makes the zynthian an easy device to use to make simple recordings confidently for technically averse people.

jofemodo, 2020-08-05:
It's implemented. Please, update and test.
Ahhh! Remember that you have to reload the webconf's dashboard to detect the pendrive presence.
It would be nice to have websocket updates for the dashboard ... perhaps some day! ;-)

closed 2020-08-05 by jofemodo

jofemodo, 2020-08-08:
Hi @wyleu !
Could you check this feature? I implemented it especially for you, mate ;-)
BTW, did you receive the AllInOne PCBs?

commit 724762dc (2020-03-11): Fix #143: Short some audio device names.