zynthian/zynthian-webconf #145: Enforce file extension for renaming captures

sehrgut, 2020-08-16:
**Is your feature request related to a problem? Please describe.**
I sometimes forget to put the file extension on when I'm renaming captures in webconf, which makes them invisible until I ssh in to rename them again.

**Describe the solution you'd like**
Change the behaviour of the rename field to only change the name portion of the filename. To ensure users are aware it will not change the extension, print the ".EXT" after the field in static text.

**Describe alternatives you've considered**
I've considered being smart enough to always type the file extension, but probably I'm going to keep forgetting. 😆 

**Additional context**
Since renaming does not trigger any sort of conversion at present, I can't think of a use case where being able to manually change the extension from webconf would be useful. Since webconf filters displayed captures by extension, the current behaviour allows users who may not know how to use ssh to hide their captures from themselves.

jofemodo, 2020-08-16:
Hi @sehrgut !
This is a perfect issue for making your first contribution to zynthian's code ... what do you think? ;-)

sehrgut, 2020-08-17:
It'll be a while before I have a computer I can run a build VM on, and I don't have a spare rpi, so I'm afraid my code contributions will have to wait for that. But I'm working on getting a zynthian build env set up.

jofemodo, 2020-08-18:
My friend, you don't need a build environment for testing this kind of changes. 
Simply fork the repository, do the changes and test on your zynthian. You can copy the modified files or clone your development fork on it. Up to you!

sehrgut, 2020-08-19:
Okay, I honestly hadn't looked into how the web stuff was generated and assumed I'd need a full build env. This does seem like a perfect getting-to-know-the-insides-of-zynthian project! I'll work on it!

mheidt, 2020-11-19:
You can test it.
I didn't show the extension  because it's a waste of space.
But I ensured that only the first part to the dot is taken from the input. The extension is taken from the original name

jofemodo, 2020-11-20:
I just tested and it works OK!
Thanks @mheidt !

closed 2020-11-20 by jofemodo