zynthian/zynthian-webconf #161: Allow disabling of unused resources (USB OTG, HDMI audio, Headphones) through web ui

mark-orion, 2021-04-03:
**Is your feature request related to a problem? Please describe.**
The current limit of a maximum number of 8 USB audio devices is further reduced by often unused ports like the headphone interface, USB OTG or HDMI audio.

**Describe the solution you'd like**
The Zynthian web ui should have options to disable these devices (done through modification of /boot/config.txt
See https://discourse.zynthian.org/t/multiple-usb-midi-issue-edit/5605/25

closed 2021-04-04 by jofemodo

jofemodo, 2021-04-04:
@riban-bw, i pushed this to stable by error.
As the commit contains a bugfix for a critical error on webconf (error 400), i would  leave it and do some extra testing instead of reverting the commit.

mark-orion, 2021-04-04:
No problem here. I just updated and checked everything. The only minor thing I can see is that there are 15 blank lines after "# Soundcard Config" in /boot/config.txt. My guess is that this should not cause any problems - so only a minor cosmetic annoyance.
Otherwise thanks for the fix and the ability to switch off resources that one does not need.

jofemodo, 2021-04-05:
OK! I just fixed this problem too. 
The audio config textarea was adding an extra new line on every "save", so you have to manually remove the extra added lines.

commit dec5f31d (2021-04-04): + Fix #161: Add options for disabling RBPi Audio & OTG, so we can get upto 3 extra audio devices.
+ Bugfix: Error 400 when saving audio settings while the soundcard select widget  is disabled.
commit 33699a7c (2021-04-05): Fix #161: Avoid adding new lines to textarea fields on every "save".