zynthian/zynthian-webconf #163: Webconf consumes all available memory

riban-bw, 2021-04-22:
**Describe the bug**
Sometimes when I click on the button in webconf to scan for plugins I get `500: Internal Server Error`. Checking the log I see no mempry available and top shows zynthian_webconf using 64.2% of system memory.

**To Reproduce**
Steps to reproduce the behaviour:
1. Within webconf navigate to SOFTWSRE->LV2 Plugins
2. Click on 'Search for new Plugins & Presets'

**Expected behaviour**
Search occurs.

**Actual behaviour**
500: Internal Server Error

**Setup:**

HARDWARE
Raspberry Pi 4 Model B Rev 1.4
Soundcard: HifiBerry DAC+ ADC PRO
Display: ZynScreen 3.5 (v1)
Wiring: MCP23017_ZynScreen
GPIO Expander: MCP23017

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05
Memory: 10% (198M/1939M)
SD Card: 39% (11G/28G)
Temperature: 42.0ºC

SOFTWARE
zyncoder: master (bb5e19c)
zynthian-ui: master (6cdb0ad)
zynthian-sys: master (3c1df40)
zynthian-data: master (194fd59)
zynthian-webconf: master (868c14a)

**Additional context**
This seems to occur when changing the content of the /zynthian/zynthian-my-data/presets/lv2/ directory and after a few checks for new plugins.

jofemodo, 2021-04-22:
I just implemented a flag for restarting webconf when needed.

jofemodo, 2021-05-03:
Hi @riban-bw !

Could we close this? I think the problem is "kind of solved" ;-)

riban-bw, 2021-05-03:
Hmmm. Kinda fixed is not what I want to hear from my doctor!

WWE have a pragmatic work-around and I think the core issue may be upstream. I am tempted to look at alternating LV2 interface but for now I will close this issue.

closed 2021-05-03 by riban-bw