zynthian/zynthian-webconf #164: List of enabled LV2 plugins not saved / restored in backup

riban-bw, 2021-05-18:
**Describe the bug**
When a backup is restored, the list of enabled LV2 plugins is not restored.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable / disable LV2 plugins
2. Backup configuration
3. Flash new image
4. Restore backup

**Expected behaviour**
LV2 list is restored.

**Actual behaviour**
LV2 list is not restored.

**Setup:**
- HARDWARE
Raspberry Pi 4 Model B Rev 1.4
Soundcard: HifiBerry DAC+ ADC PRO
Display: ZynScreen 3.5 (v1)
Wiring: MCP23017_ZynScreen
GPIO Expander: MCP23017
- SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05
Memory: 16% (314M/1939M)
SD Card: 35% (9.4G/29G)
Temperature: 45.0ºC
- SOFTWARE
zyncoder: stable (d852c8f)
zynthian-ui: stable (b2f6a7c)
zynthian-sys: stable (e8fe7ce)
zynthian-data: stable (f71259d)
zynthian-webconf: stable (153a020)

**Additional context**
This should be expected behaviour. Each time I create a new Zynthian or reflash an existing one it is a chore to reconfigure all the elements again which I believe is one of the purposes of the backup/restore.

riban-bw, 2021-05-18:
I used the data backup. I am wary of these options. It is a bit confusing. I like being able to define the level of granularity for backups but am not sure which to use and whether a system backup might restore elements I don't want, e.g. the reason I reflashed the firmware.

mheidt, 2021-11-30:
Those are the files that define the default behaviour:
/zynthian/config/system_backup_items.txt
/zynthian/config/data_backup_items.txt
Which is not part of the webconf repo.
So maybe one you two should take a look.

commit 16cabe66 (2022-01-31): Refactorize and improve Backup/Restore. It should fix #164.
closed 2022-01-31 by jofemodo