zynthian/zynthian-webconf #166: Zynthian.Local Library Snapshots -  Please Display Mode: Stage or Multi

samriccijr, 2021-07-04:
Hi All!  When Using Zynthian.local Library Snapshots:

When selecting a snapshot and viewing the programs and channels, please add a line to display the Mode as Stage or Multi.

My workaround is to rename the snapshot with the word STAGE or MULTI in it.  I have one set of controllers that prefer Stage mode and I have another (Studiologic SL880) that works best with Multi Mode.  Thank you!  Sam

mheidt, 2022-01-24:
It's already there:
SINGLE_ACTIVE_CHANNEL
A value of 1 is STAGE mode.

@jofemodo 
How do you want to have this more prominently displayed?

samriccijr, 2022-01-25:
Sometimes we cannot see the forest from the trees…. I will try to finish this.  Thank you!  Sam

closed 2024-03-12 by riban-bw