zynthian/zynthian-webconf #173: Display zs3 bank and preset in info panel

mheidt, 2022-01-02:
Like this:
[image]

closed 2022-01-02 by mheidt

mheidt, 2022-01-30:
@jofemodo 
Is it possible to merge this to stable?

jofemodo, 2022-01-31:
Yes, of course, but we should test to avoid adding unstability to stable ;-)
I would suggest to create a new "staging" branch and cherry-pick a few commits from testing that solve problems or add little, "non disruptive" features. When done, we should test carefully before merging on stable.

I will start tomorrow, OK?

Regards,

mheidt, 2022-02-06:
I guess the branch staging-2109 is not the current one you had in mind?

mheidt, 2022-02-06:
@jofemodo 
Could it be, that for Pianoteq the volume is always reset to 74%? I am not able to store the volume in a zs3 snapshot.
Has this already been fixed or should I investigate?

jofemodo, 2022-02-06:
No. It should be staging-2202 ;-)