zynthian/zynthian-webconf #176: Git Repositories and Test Checkbox

mheidt, 2022-02-13:
@jofemodo, @riban-bw 
I am still not happy with it.
It seems, that it's not possible to select all stable but one.

I would like to change it.
No Testing selectbox anymore but two buttons. "All Testing", "All Stable"
Those are changing the repositories through javascript.
The python code will only know the select boxes and no Testing checkbox anymore.

Objections?

riban-bw, 2022-02-13:
I've felt it wasn't right but not had time or inclination to look at it. Sounds like you propose something simple which is usually good. Try it and see.

closed 2022-02-13 by mheidt

jofemodo, 2022-02-14:
Hi @mheidt !

It's OK but i don't think it's a good idea to commit changes to this "staging-2203" without going testing first. Staging-2203 should be stable + a set of cherry-picked commits from testing.  If not, this is going to be really hard to manage.

Thanks!